esp8266/Arduino #9151: Error status exit 1

ardanimarwan, 2024-06-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

ardanimarwan, 2024-06-13:
FQBN: esp8266:esp8266:nodemcu
Using board 'nodemcu' from platform in folder: C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2
Using core 'esp8266' from platform in folder: C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2

"C:\\Users\\ardan\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\ardan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2/tools/mkbuildoptglobals.py" "C:\\Program Files\\Arduino IDE\\resources\\app\\lib\\backend\\resources" 10607 "C:\\Users\\ardan\\AppData\\Local\\Temp\\arduino\\sketches\\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6" "C:\\Users\\ardan\\AppData\\Local\\Temp\\arduino\\sketches\\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core/build.opt" "D:\\ARDUINO\\3. PROJECT ARDUINO\\7. IOT NODEMCU\\IOT  MODEMCU TELEGRAM\\IOT_NodeMcu_Telegram/IOT_NodeMcu_Telegram.ino.globals.h" "C:\\Users\\ardan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2\\cores\\esp8266/CommonHFile.h"
default_encoding:       cp1252
Assume aggressive 'core.a' caching enabled.
Note: optional global include file 'D:\ARDUINO\3. PROJECT ARDUINO\7. IOT NODEMCU\IOT  MODEMCU TELEGRAM\IOT_NodeMcu_Telegram\IOT_NodeMcu_Telegram.ino.globals.h' does not exist.
  Read more at https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html
Detecting libraries used...
C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core/build.opt -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core -c @C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_NODEMCU_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU_ESP12" -DARDUINO_BOARD_ID="nodemcu" -DFLASHMODE_QIO -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\nodemcu C:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6\sketch\IOT_NodeMcu_Telegram.ino.cpp -o nul
Alternatives for CTBot.h: [CTBot@2.1.13]
ResolveLibrary(CTBot.h)
  -> candidates: [CTBot@2.1.13]
C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core/build.opt -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core -c @C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_NODEMCU_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU_ESP12" -DARDUINO_BOARD_ID="nodemcu" -DFLASHMODE_QIO -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\nodemcu -Ic:\Users\ardan\OneDrive\Documents\Arduino\libraries\CTBot\src C:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6\sketch\IOT_NodeMcu_Telegram.ino.cpp -o nul
Alternatives for ArduinoJson.h: [ArduinoJson@7.0.4]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@7.0.4]
C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\ardan\AppData\Local\Temp\arduino\sketches\6AAC056A8E6B1A8C5FDCF0B18E7E5DF6/core/build.opt -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\ardan\Ap [... cut ...]
      |         ^~~~~~~~~~
cc1plus.exe: some warnings being treated as errors

Using library CTBot at version 2.1.13 in folder: C:\Users\ardan\OneDrive\Documents\Arduino\libraries\CTBot 
Using library ArduinoJson at version 7.0.4 in folder: C:\Users\ardan\OneDrive\Documents\Arduino\libraries\ArduinoJson 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\ardan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi 
exit status 1

Compilation error: exit status 1

mcspr, 2024-06-13:
Please do not open issues related to libraries unless it is really an issue with our Core. CTBot author should add support for ArduinoJson v7, just use ArduinoJson v5/v6 in the meantime

https://github.com/shurillu/CTBot/issues is the correct repo for the issue

https://github.com/search?q=repo%3Ashurillu%2FCTBot%20ARDUINOJSON_VERSION_MAJOR&type=code only shows support for v5 and v6

> Alternatives for ArduinoJson.h: [ArduinoJson@7.0.4]
> ResolveLibrary(ArduinoJson.h)
> -> candidates: [ArduinoJson@7.0.4]

> c:\Users\ardan\OneDrive\Documents\Arduino\libraries\CTBot\src\CTBotInlineKeyboard.cpp: In member function 'bool CTBotInlineKeyboard::addButton(const String&, const String&, CTBotInlineKeyboardButtonType)':
c:\Users\ardan\OneDrive\Documents\Arduino\libraries\CTBot\src\CTBotInlineKeyboard.cpp:75:2: error: 'button' was not declared in this scope
75 | button["text"] = URLEncodeMessage(text);
| ^~~~~~

https://github.com/shurillu/CTBot/blob/a0d7f5fe8a471a8fd2f9b28fc936a66775f612cf/src/CTBotInlineKeyboard.cpp#L69-L72

closed 2024-06-13 by mcspr