esp8266/Arduino #9154: elf2bin.py: Verify/compile outputs two syntax warnings

Vic-Y, 2024-06-18:
### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform
- Hardware:					[ESP8266 LOLIN(WEMOS) D1 mini (clone)]
- Core Version:
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE
- Module:  			[LOLIN(WEMOS) D1 mini (clone)]
- Flash Mode:			[DOUT(compatible)]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
Output window displays the following two syntax warnings during verify/compile:
 
/home/{user}/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/elf2bin.py:54: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)
/home/{user}/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/elf2bin.py:73: SyntaxWarning: invalid escape sequence '\s'
  words = re.split('\s+', line)

mcspr, 2024-06-19:
> I guess that the escape sequence '\s' should be replace by a space character, but since I do not know how elf2bin.py file works, I did not submit a commit.
words = re.split(' +', line)

Fixed in master by https://github.com/esp8266/Arduino/pull/9034, should be part of the next release

closed 2024-06-19 by mcspr

madurapa, 2025-02-20:
You can use `dev` in the version to fix this warning (note this may break at any time)

```
esp8266:
  board: nodemcuv2
  framework:
    version: dev
```
More information on https://esphome.io/components/esp8266.html