esp8266/Arduino #9156: ESP8266 NodeMCU SD Card Not Initializing

jcrosby10, 2024-06-20:
I have tried numerous tutorials, Arduino examples and I cant get an SD to initialize. There is a card inserted that is formatted with exFAT and there is one file test.txt. Particularly [SD Card Module With ESP8266][1] and [ESP8266 (NodeMCU) and SD Cards][2].

I'm using this SD card module, [Micro SD SDHC TF Card Adapter Reader Module with SPI Interface Level Conversion Chip Compatible with Arduino Raspberry PI 10pcs][3]. I soldiered the pin connector to the SD card so it would easily set in the breadboard.

Here are a few pics of my breadboard setup:

[![Breadboard setup#1][4]][4]

[![Breadboard setup#2][5]][5]

[![Breadboard setup#3][6]][6]

    void SDsetup()
    {
    pinMode(10, OUTPUT);
    digitalWrite(10, HIGH);

    File Textfile;
    Serial.begin(9600);
    Serial.println("Initializing SD card");

    if (!SD.begin(15)) {  
        Serial.println("Initialization failed!");  
        return;
    }

    Serial.println("Initialization completed"); 
    Textfile = SD.open("test.txt", FILE_WRITE);  

    if (Textfile) {
        Serial.println("Writing to Textfile...");  
        Textfile.println("First line");    
        Textfile.println("1, 2, 3, 4, 5");
        Textfile.println("a, b, c, d, e");
        Textfile.println();
        Textfile.close();     
        Serial.println("Completed");       
        Serial.println();
    }
    else {
        Serial.println("Text file could not be read");  
    }

    // READ TEXTFILE
    Textfile = SD.open("test.txt");      
    if (Textfile) {
        Serial.println("test.txt:");    
        while (Textfile.available()) {
        Serial.write(Textfile.read());     
        }
        Textfile.close();    
    }
    else  {
        Serial.println("Text file could not be opened"); 
    }
    }

The serial output:

```none
smart setup
Initializing SD card
Initialization failed!
smart loop
```

  [1]: https://www.instructables.com/SD-Card-Module-With-ESP8266/
  [2]: https://pawitp.medium.com/esp8266-nodemcu-and-sd-cards-55e0d073d959
  [3]: https://www.amazon.com/dp/B0989SM146
  [4]: https://i.sstatic.net/kEew5i5b.jpg "Breadboard setup#1"
  [5]: https://i.sstatic.net/CbGObifr.jpg "Breadboard setup#2"
  [6]: https://i.sstatic.net/TQd4BrJj.jpg "Breadboard setup#3"

I found this https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#file-system-object-spiffs-littlefs-sd-sdfs and surprisingly SPIFFS sort of worked. It initialized but had problems with read/write.

    File Textfile;
    Serial.begin(9600);
    Serial.println("Initializing SD card");

    if (SPIFFS.begin()) {
        Serial.println("Siffs success");
        Textfile = SPIFFS.open("test.txt", "w");
        if (Textfile) {
            Serial.write(Textfile.read());
            Serial.println("Writing to Textfile...");
            Textfile.println("First line");
            Textfile.println("1, 2, 3, 4, 5");
            Textfile.println("a, b, c, d, e");
            Textfile.println();
            Textfile.close();
            Serial.println("Completed");
            Serial.println();
            Serial.write(Textfile.read());
        }
    }
    else {
        Serial.println("Text file could not be read");
    }
    if (!SD.begin(15)) {
        Serial.println("Initialization failed!");
        return;
    }

The above code produced the below log:

> Initializing SD card<br>
> Siffs success<br>
> �Writing to Textfile...<br>
> Completed<br>
> <br>
> �Initialization failed!

Why cant I get either one to work properly? What am I missing?

mcspr, 2024-06-20:
> There is a card inserted that is formatted with exFAT and there is one file test.txt

exFAT is not currently supported by our APIs, only FAT

We would have to change some things internally, for instance https://github.com/esp8266/Arduino/blob/7e0d20e2b9034994f573a236364e0aef17fd66de/libraries/SDFS/src/SDFS.h does explicitly name only FAT FS and File objects.
Per https://github.com/earlephilhower/ESP8266SdFat/tree/master/src, this should be changed to FsLib API instead, which shares support for both FAT and exFAT

jcrosby10, 2024-06-20:
Yes I just found that out. I was referred to sdcard.org to download a formatter since Windows will only let me do exFAT and NTFS.