esp8266/Arduino #9161: Releasing Updates

ayushsharma82, 2024-07-09:
Pretty sure you guys have already thought of this but just an opinion - Last release was more than a year ago. I think you should start releasing minor versions of this core with recent bug fixes and feature requests?

A OSS library like mine (NetWizard) is already waiting for the removable routes and webserver filters which were added recently ( but the update for core is yet to be released ), without the release I cannot add ESP8266 support.

nopnop2002, 2024-07-16:
This is just my personal opinion, but I think EspressIF spends a lot of time supporting esp-idf.

JAndrassy, 2024-07-16:
@nopnop2002 this repository is not maintained by Espressif

nopnop2002, 2024-07-16:
@JAndrassy 

i had a misunderstanding

ayushsharma82, 2024-08-09:
@mcspr  A beta release for now so that community can test and give us feedback before finally releasing all the fixes?

mcspr, 2024-08-09:
> @mcspr A beta release for now so that community can test and give us feedback before finally releasing all the fixes?

Git install is not an option? Both ArduinoIDE and PIO seamlessly handle it
https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

Let me figure out some remaining things though, and 3.2.0 would be published proper.

If git is out of the question... running build package .sh seems to produce some relevant files. I would guess package index .json should work in IDE
https://github.com/mcspr/esp8266-Arduino/releases/tag/3.2.0-dev
(but, idk if it actually works. open an issue there instead if it does not)

ayushsharma82, 2024-08-09:
@mcspr For a layman, no, git install is too tedious. Surely we both can do it but most users won't unless strictly necessary. Most of the users are happy to use the board manager in Arduino IDE and PIO because it handles all the steps for them.

Also, tagged releases are important, this ensures the user that a specific tagged release is expected to work properly without critical bugs.

I know you guys have been actively merging PRs, all this can be done more efficiently within this repo. Take a look at appwrite’s repository and how they use Github projects to categorize issues and manage future releases. Ref: https://github.com/orgs/appwrite/projects/8/views/1?sliceBy%5Bvalue%5D=x.x.10

atesin, 2025-04-18:
> Git install is not an option? Both ArduinoIDE and PIO seamlessly handle it
> https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

sure it may, but that will force everybody us normal users to turn into core developers, for devs those procedures may be trivial daily steps and chances are high they have already installed git, python, platformio, vscode, eclipse, gcc, esptool, android studio, etc etc etc  ...  but for most of us we started to feel forced to install some tools unknown just to run once a year, and as we have no advanced git/toolchain knowledge chances we end messing up something are high

as we advance our esp8266-arduino development skills (normal sketches) we reach some weird behavior sometimes that we have no idea why occur... and last release is almost 2 years now!

please, think of us