esp8266/Arduino #9162: Request for Help with Implementing MAC Filter on ESP8266 in AP Mode using Arduino Platform(PlatformIO)

SajjadQadri, 2024-07-10:
Hello Everybody,

I am working on a project using the ESP8266 and need to implement a MAC filter in Access Point (AP) mode. I am using the Arduino platform(PlatformIO) for programming and require your assistance.

My main issue is how to define a list of allowed MAC addresses and disconnect a client if its MAC address is not on the allowed list.

I have tried several different codes so far, but it seems that some of these codes do not work correctly. For example, the code that was previously suggested to me:
void wifi_station_deauth(uint8_t *bssid) {
struct auth_deauth_frame {
uint16_t frame_control;
uint16_t duration;
uint8_t da[6];
uint8_t sa[6];
uint8_t bssid[6];
uint16_t seq_ctrl;
uint16_t reason_code;
} __attribute__((packed));

struct auth_deauth_frame frame;
frame.frame_control = 0x00C0;
frame.duration = 0;
memcpy(frame.da, bssid, 6);
memcpy(frame.sa, WiFi.softAPmacAddress().c_str(), 6);
memcpy(frame.bssid, WiFi.softAPmacAddress().c_str(), 6);
frame.seq_ctrl = 0;
frame.reason_code = 0x0001; // Unspecified reason

wifi_send_pkt_freedom((uint8_t *)&frame, sizeof(frame), 0);
}

This code does not disconnect clients whose MAC addresses are not on the allowed list. I have been using standard Arduino library functions as well as lower-level SDK functions, but I have not been able to resolve the issue.

Has anyone had a similar experience or can provide a comprehensive guide for creating an effective MAC filter on the ESP8266? Any sample code or suggestions in this regard would be greatly appreciated.

Thank you in advance for your help and guidance.

Best regards,
Sajjad Qadri

enjoyneering, 2024-07-11:
Hi,

I'm not a C/C++ expert, but I [read](https://www.gnu.org/software/c-intro-and-ref/manual/html_node/Packed-Structures.html) that it is not recommended to use pointers with packed structures.

SajjadQadri, 2024-07-13:
hi enjoyneering
So what is your suggestion?