esp8266/Arduino #9164: SoftAP appears to default to TKIP.

Frostie314159, 2024-07-13:
## Note
This issue is transferred from WLED https://github.com/Aircoookie/WLED/issues/4052 I'm not in posession of the hardware, which is required for testing this. I found the issue through a packet capture and reported it at WLED, where the maintainers noted, that it's an issue in this library. I filled out the fields to my best knowledge, but this is definitely a software issue, across all boards. 
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [?] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					Unknown
- Core Version:			4.2.1
- Development Env:		PlatformIO
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			Any
- Flash Mode:	Any
- Flash Size: 		Any
- lwip Variant:   Any
- Reset Method:		Any
- Flash Frequency:	Any
- CPU Frequency:		Any
- Upload Using:		Any
- Upload Speed:   Any

### Problem Description

When operating a SoftAP initialized With `WiFi::SoftAP`, the encryption defaults to TKIP aka WPA1, which has been deprecated for over a decade by now, due to it being insecure. Alternatively, CCMP aka WPA2 should be used.
I'm aware, that this might be for backwards compatibility, but if TKIP is used, a warning should be emitted. 
If WPA/WPA2 is used, this still presents a risk, because TKIP is implicitly used as the group cipher, which makes all group addressed traffic vulnerable.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
See minimal WiFi SoftAP example.

mcspr, 2024-07-17:
We have a pretty limited control over the WiFi internals, driver is inside a .a blob coming from the NONOS SDK
https://github.com/espressif/ESP8266_NONOS_SDK
https://github.com/esp8266/Arduino/tree/master/tools/sdk/lib

It *may* be an option for RTOS variant, but it is an entirely different can of worms
https://github.com/espressif/ESP8266_RTOS_SDK/

If the intent is to exclude TKIP from group and pairwise config, these lines *seem* to affect it. No idea whether it works, though. You can attempt to build a basic WiFi AP example from the SDK and check whether that's true or false
https://github.com/espressif/ESP8266_RTOS_SDK/blob/d3a5f99ef8acecb3e0fbba8493aba15fa50fe375/components/wpa_supplicant/src/esp_supplicant/esp_hostap.c#L67

Frostie314159, 2024-07-17:
I'm on vacation right now and don't have access to my laptop, however from what I can see, this is an issue with the rtos sdk forcing WPA/WPA2. The thing is, that I'm unfamiliar with nonos and just discovered this from the outside. I'm going to open an issue on the rtos sdk, but I'm not sure where that'll go, since the last release was three years ago. 
I'm part of the project to reverse engineer the esp32 wifi PHY and MAC, and in my experience the Espressif blobs do offer configurability in that.