esp8266/Arduino #9169: ESP8266Webserver Example with gzip Files

hasenradball, 2024-07-17:
Hello ESP Team,

I wanted to create an webserver on an ESP8266 and serve compressed files (e.g.: index.html.gz) instead of uncompressed files.
Is there an example how to setup/handle this?

mcspr, 2024-07-17:
FSBrowser has a minimal example.

The gist is setting `Content-Encoding` header so the browser decodes the gzip'ed stream into html, but keep `Content-Type: text/html`
https://github.com/esp8266/Arduino/blob/7e0d20e2b9034994f573a236364e0aef17fd66de/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino#L458-L473

It serves a static array. In case of filesystem, mime-type match based on the file extension happens.
https://github.com/esp8266/Arduino/blob/7e0d20e2b9034994f573a236364e0aef17fd66de/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L254

Our internal handlers for `static` file serving automatically change encoding header, so just send out the file response w/ the correct type
https://github.com/esp8266/Arduino/blob/7e0d20e2b9034994f573a236364e0aef17fd66de/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h#L646-L657

closed 2024-07-17 by mcspr

hasenradball, 2024-07-18:
@mcspr Dear Max,

thank you for your reply, so as I understood, I have to use and overload function of send together with the corresponding header.

Frank