esp8266/Arduino #9195: Serial.available()  have a issue, its not able to return availiable bytes

The3ven, 2024-08-29:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-07]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu v1.0 (ESP12E Module) ]
- Flash Mode:			[Not a Option]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[Not a Option]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Nodemcu `Serial.available();` function is not able to send total bytes available on serial it just send 1.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include "Arduino.h"

void setup()
{
    Serial.begin(9600);
}

String tmp = "";

void loop()
{
    while (Serial.available() > 0)
    {
        Serial.println("data in buffer : ");
        Serial.print(Serial.available());
        Serial.println("\n");
        tmp = Serial.readString();
    }

    if (tmp.length() > 4)
    {
        Serial.println("Data : " + tmp);
        Serial.println("Data length : " + String(tmp.length()));
        Serial.flush();
        tmp = "";
    }
}

```

### Debug Messages

```
⸮R⸮⸮⸮⸮data in buffer : 
1

Data : hi hi hi

Data length : 10
⸮R⸮⸮⸮⸮R⸮⸮fgFj⸮data in buffer : 
1

Data : hihihih

Data length : 9

```

mcspr, 2024-08-30:
> function is not able to send total bytes available on serial it just send 1.

does exactly what you have asked, though?
`tmp = readString()` **replaces** the whole string, perhaps you meant `+=`

closed 2024-08-30 by mcspr

The3ven, 2025-06-04:
> > function is not able to send total bytes available on serial it just send 1.
> 
> does exactly what you have asked, though? `tmp = readString()` **replaces** the whole string, perhaps you meant `+=`

why should i append it?

i have to know all the available data, readString() from serial send all the data which is available on serial at the present time and empty the serial buffer, 

Now back to the available part before i get all the data from serial i wan to know the length and it says one but it got read 9 characters from the buffer, 

How??

if available told me there is only 1 character then how it read 9 characters from buffer??