esp8266/Arduino #9197: [ESP8266 - Timezones]: TZ.h has wrong definitions

hasenradball, 2024-09-01:
Dear ESP Team, @mcspr ,

I used the `TZ.h` definitions in a project and realized that some things going wrong... 
Take the following picture as a example:

[image]

If I feed the Timezome like `UTC+2` or `GMT+2` I expected to get the Time from Berlin (in Summer).
Lets assume we have `UTC0` = 12:00 and i feed `UTC+2` I get the time 10:00 but expected 14:00.

So I think the is a error in the definitions.

Is it?

I used the latest `TZ.h`

best Frank

mcspr, 2024-09-02:
You'd have to take a look at TZ variable definition and what 'offset' is:
https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html

> The offset specifies the time value you must add to the local time to get a Coordinated Universal Time value. It has syntax like [+|-]hh[:mm[:ss]]. This is positive if the local time zone is west of the Prime Meridian and negative if it is east. The hour must be between 0 and 24, and the minute and seconds between 0 and 59.

Berlin is on the east of the Prime Meridian (offset +00:00), hence you'd use negative offset and not positive.

> If I feed the Timezome like UTC+2 or GMT+2 I expected to get the Time from Berlin (in Summer).

Use `TZ_Europe_Berlin`?
https://github.com/esp8266/Arduino/blob/ccea72823ac50290bc05c67350d2be6626e65547/cores/esp8266/TZ.h#L440

closed 2024-09-02 by mcspr

hasenradball, 2024-09-02:
Thanks for the Information,

so it is a misunderstanding from my point of view.

I created a wordclock with the ESP8266 and transferrred the TZ.h in json format to easily set the TZ on website.

[image]

Therfore I thought `GMTp3` means GMT+ 3 Hours to East.

A short question al least.

The TZ.h is very long, do you think it is possible to easily reduce the lines to unique entries?
It seems there are many double entries in side.

mcspr, 2024-09-02:
> I created a wordclock with the ESP8266 and transferrred the TZ.h in json format to easily set the TZ on website.
> The TZ.h is very long, do you think it is possible to easily reduce the lines to unique entries?
It seems there are many double entries in side.

I believe it would be much easier to make it local to the project, see
https://github.com/esp8266/Arduino/blob/master/tools/format_tzdata.py

Entries here repeat tzdata verbatim, plus I'd expect regional names are used instead of hard numbered offsets. Etc zone is a weird one, though, that I agree.

hasenradball, 2024-09-02:
What I did is created a json object in js to be able to select it via drop down.

For this it is normally no need for the TZ.h on the ESP side.

What do you mean by keep it locally to the project?

mcspr, 2024-09-02:
> What I did is created a json object in js to be able to select it via drop down.
> 
> For this it is normally no need for the TZ.h on the ESP side.
> 
> What do you mean by keep it locally to the project?

To make tzdata the source of timezone data, not TZ.h. You can modify the script to generate the json  like it generates either TZ.h or markdown.

hasenradball, 2024-09-02:
Ah now I got your point…

I thought of this to modify the python script but with a few mouse clicks I easily got the json from TZ.h.

So no urgent need to change or modify the python.