esp8266/Arduino #9202: ESP8266Wifi Header not Found

Grasstown, 2024-10-25:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01(S)
- Core Version:			[3.1.12:03/20/2023]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

Before I start, this is my first github issue so I'm not too sure on etiquette and I hope I filled out everything correctly. When going through the Wiki examples for connecting to wifi, I get a compilation error saying the ESP8266Wifi header could not be found. I've tried both types of installs (through the board manager and installing from git) and I get the same error message. Do I need to directly copy the libraries folder of this repo into my Arduino15 / sketch directory?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266Wifi.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
C:\Users\kevin\Documents\Arduino\esp8266test\esp8266test.ino:1:10: fatal error: ESP8266Wifi.h: No such file or directory
    1 | #include <ESP8266Wifi.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Alternatives for ESP8266Wifi.h: []
ResolveLibrary(ESP8266Wifi.h)
  -> candidates: []
exit status 1

Compilation error: ESP8266Wifi.h: No such file or directory
```

mcspr, 2024-10-30:
> When going through the Wiki examples for connecting to wifi, I get a compilation error saying the ESP8266Wifi header could not be found. I've tried both types of installs (through the board manager and installing from git) and I get the same error message.

Re-check that you are actually selecting the ESP8266 board when building the sketch!

>  Do I need to directly copy the libraries folder of this repo into my Arduino15 / sketch directory?

Library is included as part of the ESP8266 Arduino Core and should be resolved automatically; no manual c/p required, no need for any additional libraries from the IDE installer UI

closed 2024-10-30 by mcspr

mcspr, 2024-10-30:
Also... it is ESP8266WiFi.h not ESP8266Wifi.h

Grasstown, 2024-10-30:
Thanks! I can't believe I made such a dumb issue... That worked.