esp8266/Arduino #9205: NTP-TZ-DST esp8266 example displays year "117"

atesin, 2024-11-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			nodemcu v3 lolin (with esp8266ex)
- Core Version:			3.1.2
- Development Env:		arduino ide 2.3.3
- Operating System:		windows 10
### Settings in IDE

- Module:  			nodemcu 1.0
- Flash Mode:			i dont know
- Flash Size: 		4mb
- lwip Variant:   v2 low memory
- Reset Method:		dont know
- Flash Frequency:	dont know
- CPU Frequency:		80mhz
- Upload Using:		serial
- Upload Speed:  115200 baud 8n1 no flow control, send line mode, `<LF>` endlines

### Problem Description

- ntp is not even mentioned time in readthedocs documentation (btw the search feature is broken so i had to look the entire site)
- but at least i see there is an included example NTP-TZ-DST
- the example is very full featured, but very complicated and hard to understand too, could had been better splitted in many specific "minimal" ones (there are a lot of complaints everywhere about this)
- there are no clear info about sntp library functions, i just want to sync my `time()` function correctly
- does `configTime(MYTZ, "pool.ntp.org");` saves configuration in flash? because if it does, which could be a safe way to get ntp time without wearing out the flash each time?
- when i run the example it displays year "124", it is shifted 1900 years back (ntp protocol epoch is 1900, unlike unix epoch)
- is not enough with adding 1900 to year, there are a lot of further related calculations like: days of week, leap years, moon phases, sunrise/sunset hours, star position, equinox/solstices, and many other that require correct full dates
- i remember some avr arduino time libraries that uses avr libc time.h constans to correct these values on [set_system_time()](https://www.nongnu.org/avr-libc/user-manual/group__avr__time.html#ga02663b926feab893c0e0ec1f29b3c315) before and __immediately after__ got ntp time like [NTP_OFFSET](https://www.nongnu.org/avr-libc/user-manual/group__avr__time.html#gaa7c3e93bff90674e9e74de6bdec85613)

```
set_system_time(ntp_timestamp - NTP_OFFSET);
```

- is this project abandoned?, because forum is closed (nobody anwers esp32.com subforum), there is no new commits in months, last stable release is about 1.5 years old, this project is based in NONOS sdk that was obsoleted time ago by espressif in favor of RTOS one.... please dont conplain me about posting here because all support channels are dying

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// just the full NTP-TZ-DST example sketch shipped with esp8266-arduino project version 3.1.2

```

### Debug Messages

```
localtime: isdst=0 yday=316 wday=1 year=117 mon=10 mday=13 hour=17 min=8 sec=2
gmtime:    isdst=0 yday=316 wday=1 year=117 mon=10 mday=13 hour=17 min=8 sec=2
clock:     60s + 77083000ns
millis:    60077
micros:    60077116
gtod:      1510592882s + 999354us
time:      1510592882
timezone:  GMT0BST,M3.5.0/1,M10.5.0
ctime:     Mon Nov 13 17:08:02 2017

time(): 1510592883   gettimeofday(): 1510592883.971491  seconds are unchanged
time(): 1510592884   gettimeofday(): 1510592884.021561  <-- seconds have changed
```

atesin, 2024-11-17:
i answer myself (after rolling over whole internet and waking up a lot of people...):
```
◆ tm_year
int16_t tm::tm_year

years since 1900
```

- https://www.nongnu.org/avr-libc/user-manual/structtm.html#a933e733942822b2def4aa966ee811293

so... the "124" means that PRESENTLY, HAD BEEN PASSED "124" `years since 1900` what is OK... 

i was printing a `tm` struct fields directly.. it seem that the `tm` structs are not intended to be used directly (or at least use with care) .... same with other numbers that start with "0" instead "1" (i.e. tm_mon 0 = january, etc)

 for printing better use [ctime()](https://www.nongnu.org/avr-libc/user-manual/group__avr__time.html#ga72a5383fadfbaa033ac36e9e45b0bd6d) or [strftime()](https://en.cppreference.com/w/c/chrono/strftime) instead

- [<time.h> - avr libc (new page)](https://avrdudes.github.io/avr-libc/avr-libc-user-manual/group__avr__time.html)
- [<tilme.h> - cppreference](https://en.cppreference.com/w/c/chrono)

mcspr, 2024-11-17:
> it seem that the tm structs are not intended to be used directly (or at least use with care)

See POSIX ctime reference, this is not something specific to either avr or esp8266 libc
https://pubs.opengroup.org/onlinepubs/9799919799/

> does configTime() writes configuration to flash?, and if does, what is the safer way to sync periodically without tearing flash?

No. Why would it? Where you got that idea from?

> could anybody please split NTP-TZ-DST example into various smaller basic minimal ones to explain separate features?? (no intermediate functions, no macros, no spaghetti codes)

You are welcome to update the example(s)

> could anybody please write a doxygen-like esp* sntp functions documentation thad nobody exactly knows?

See lwip documentation for the useful part. This is mentioned in the readme btw. SDK methods are not guaranteed to work correctly, please do not refer to NONOS examples or source using NONOS directly
https://www.nongnu.org/lwip/2_1_x/sntp_8h.html

closed 2024-11-17 by mcspr

atesin, 2024-11-19:
> See POSIX ctime reference, this is not something specific to either avr or esp8266 libc
> https://pubs.opengroup.org/onlinepubs/9799919799/

what is that?... i see info on these pages:
- https://avrdudes.github.io/avr-libc/avr-libc-user-manual/modules.html (arduino avr compatibility)
- https://en.cppreference.com/w/c/chrono/tm (c/c++ reference maintained by enthusiasts)

> No. Why would it? Where you got that idea from?

i think it prefectly could save config to flash with `configTime(ts, srv1, srv2, srv3)` and then sync time with `configTime()` just using stored config, like `WiFi.begin()` does

> could anybody please write a doxygen-like esp* sntp functions documentation thad nobody exactly knows?

you are my hero! ...  however it seem lwip sntp functions cannot be used directly in sketches (i tested and compiler complain me about `sntp_init() not defined`) ...  i would like to found some documentation about esp8266-arduino SNTP CORE functions (those used in NTP-TZ-DST without including any header, except `<TZ.h>` maybe)

mcspr, 2024-11-19:
sub-frames are to blame...
https://pubs.opengroup.org/onlinepubs/9799919799/basedefs/time.h.html
tm struct is both in c standard, and posix. my point here is this is not something platform-specific or unique to esp8266 implementation. cppreference page might've been a better link, as it mentions c standard.

> i think it prefectly could save config to flash with configTime(ts, srv1, srv2, srv3) and then sync time with configTime() just using stored config, like WiFi.begin() does

not the case, and never mentioned in the example [or the configTime() source](https://github.com/esp8266/Arduino/blob/ccea72823ac50290bc05c67350d2be6626e65547/cores/esp8266/time.cpp#L232-L242))
its a helper function that does [`setenv("TZ", ...); tzset();`](https://pubs.opengroup.org/onlinepubs/9799919799/functions/tzset.html) and then sets up sntp servers in a single call. TZ is a runtime value, independent of the sntp. SNTP servers are also runtime values, independent of TZ. It is up to the user to set up these strings, either hard-coded or retrieved from some kind of dynamic config store

> i would like to found some documentation about esp8266-arduino SNTP CORE functions (those used in NTP-TZ-DST without including any header, except <TZ.h> maybe)

`#include <sntp.h>` does that, right at the top. without the include compiler would now know how to call those functions, just the way c and c++ work. dry symbol documentation can be found at the lwip site, example gives a possible use-case in the code directly

> i am not a programmer but a sysadmin, i consider myself a self-learner guy, so sometimes things got harder to me... i am fairly new in programming and C and there are many things i dont know  (i.e. using git, do debug, etc) and lot of things i dont even imagine i am sure xD

We cannot presume how much user does or does not know about the programming environment, and pretty sure these examples were never intended to be used as isolated learning exercises. API usage example, yes. Semi-interactive documentation, probably.