esp8266/Arduino #9211: Memory corruption by wifi_set_opmode_current()

neuweiler, 2024-12-08:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			3.1.2
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 R1
- Flash Mode:			Arduino IDE default
- MMU:            32kB cache + 32kB IRAM
- non-32-bit access: use pgm_read macros
- VTables:       IRAM
- Flash Size: 		4MB/1MB
- lwip Variant:   v2 Higher Bandwidth
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

A call of the function wifi_set_opmode_current() causes partial memory corruption. 
In my code I have a class Config which holds the configuration loaded from a JSON file. When initializing the ESP8266 to act as an AccessPoint (AP), I call WiFi.softAPConfig() or WiFi.softAP() which both call ESP8266WiFiGenericClass::enableAP() and mode() which results in a call to wifi_set_opmode_current(). By copying the code of ESP8266WiFiGenericClass::mode() to my WLAN class' initialization, I was able to pinpoint where part of the Config memory gets overwritten.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

    logger.debug("a pw: %s ssid: %s, ch: %d", config.wifiApPassword, config.wifiApSsid, config.wifiApChannel);
    bool ret = false;
    ETS_UART_INTR_DISABLE();
    ret = wifi_set_opmode_current(m);
    ETS_UART_INTR_ENABLE();
    logger.debug("b pw: %s ssid: %s, ch: %d", config.wifiApPassword, config.wifiApSsid, config.wifiApChannel);

```

### Debug Messages

```
5951 - DEBUG: a pw: inverter ssid: Solar, ch: 7
mode : sta(2c:3a:e8:23:3f:cf) + softAP(2e:3a:e8:23:3f:cf)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
5963 - DEBUG: b pw:  ssid: ��������������������, ch: 7
6171 - INFO: started WiFi AP �������������������� on ip 192.168.4.1, channel 7
```

As you can see, both the ssid and the password in the class Config get overwritten. A simmilar issue was recorded here: https://github.com/espressif/ESP8266_RTOS_SDK/issues/117

Code in above state with debugging and copied code from mode() is available on https://github.com/neuweiler/SolarInverterToWeb/tree/0.9 . You will have to upload the files in /data with the LittleFS Uploader tool. Either add a working SSID/Password for your own AP or wait 30sec after reboot (countdown reaches 0)

mcspr, 2024-12-08:
Have you seen this issue with WiFi::mode, and you are attempting to reproduce it using your local class?
Have you tried a smaller example, then? Also, a minor note. Unlike WiFi class, you are not waiting for SDK methods to complete (fpm)

Lines above mention 'wifiStationSsid', and here is the line assigning it
https://github.com/neuweiler/SolarInverterToWeb/blob/30965e673b683f381f22f54a9c19c447351d5b9e/Config.cpp#L57

Pointer is assigned from StaticJsonDocument, which is created on the stack.
```cpp
void Config::load() {
    File file = LittleFS.open(CONFIG_FILE, "r");
    if (!file) {
        logger.error("Failed to open %s", CONFIG_FILE);
    }

    StaticJsonDocument < 2048 > doc;
    DeserializationError error = deserializeJson(doc, file);
...
    wifiStationSsid = doc[F("wifi")][F("station")][F("ssid")] | "";
...
```

Pretty sure whatever stack memory that gets used does not survive between the functions, and by logging the result you simply overwrite at some point, especially calling the Logger / printfs

closed 2024-12-08 by mcspr

neuweiler, 2024-12-09:
My apologies, you're right! The config fields are just references to the jsonDoc which was a) too big for the stack and b) only had a live-time during intitialization of the Config class. I wasn't aware that StaticJsonDoc was placed on the stack - but the docs clearly state it. Fixed now (on master branch) and calling wifi_set_opmode_current() no longer causes any problems. The solution works like a charm now, thanks.