esp8266/Arduino #9217: Compilation error - python3.exe: The parameter is incorrect

Total-Reality, 2025-01-10:
#### Platform

- Hardware:				ESP8266 ESP-12F WiFi - Node MCU
- Development Env:		Arduino IDE 2.3.4
- Operating System:		Windows 11

### Settings in IDE

- Module:  			ESP8266 > NodeMCU 1.0 (ESP-12E Module)

### Problem Description

**When checking/compiling, the following error occurs immediately:**

fork/exec C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3.exe: The parameter is incorrect.

Compilation error: fork/exec C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3.exe: The parameter is incorrect.

This problem did not exist in Arduino IDE 2.1.1. It occurs in version 2.3.4 and in version 2.2.1.

It occurs with every .ino file. Even if it is only equipped with an empty setup/loop.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

mcspr, 2025-01-10:
Can you set log in the IDE to be more verbose, so it actually shows the script in question?
Is it portable IDE (.zip), Microsoft Store, or globally installed one (.exe / .msi installer)?
python3.exe actually works, in console / Win+R, there is an interactive shell launched?

Previously reported at https://github.com/esp8266/Arduino/issues/8945, but not actually resolved

Total-Reality, 2025-01-11:
> Can you set log in the IDE to be more verbose, so it actually shows the script in question? Is it portable IDE (.zip), Microsoft Store, or globally installed one (.exe / .msi installer)? python3.exe [...]

Thanks for your quick reply.
It is installed globally with an EXE file.

I found out that the problem, because of your answer in #8945 
The reason is a feature of Comodo Firewall (Auto-Containment). 

The program blocks/isolates executable files that are indirectly executed by other files. Normally the program asks whether an isolated file should be executed or further isolated. In this case the program may not give a message because it is taking place within Sketch.
It is also strange that the problem did not exist in version 2.1.1.

The problem is solved when I add an exception for the Arduino Folder in Comodo.
So it is probably not a bug, but an incompatibility.