esp8266/Arduino #9218: Some valid SSL certificates crash (BearSSL) ("ssl_hs_client.c" causes crash in "pgmspace.h")

Rob58329, 2025-01-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

Platform

    Hardware: ESP8266 Wemos D1 Mini
    Core Version: [github as at 30/12/24]
    Development Env: [Arduino IDE v1.8.19]
    Operating System: [Windows 10]

Settings in IDE

    Module: [Wemos D1 mini]
    Flash Mode: [?]
    Flash Size: [4MB]
    lwip Variant: [v2 Lower Memory]
    Reset Method: [?]
    Flash Frequency: [?]
    CPU Frequency: [80Mhz]
    Upload Using: [SERIAL]
    Upload Speed: [115200]

### Problem Description

The below sketch/specific-SSL certificate crashes the ESP8266 on "client.connect()" (every time).

Note that if you invalidate the below SSL certificate (by changing a few characters) the "client.connect()" fails but the ESP8266 does NOT crash.

Also note that using an earlier github (such as 1-June-2021) works fine (ie the below sketch connects fine and does not crash).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#define smtp_address "smtp.hosts.co.uk"
#define smtp_port_secure 465

#include <ESP8266WiFi.h>
#include <time.h>

char *stack_start;
uint32_t stack_size() {char stack; return (uint32_t)stack_start - (uint32_t)&stack;}

#ifndef STASSID
  #define STASSID "ssid"
  #define STAPSK  "password"
#endif
const char *ssid = STASSID;
const char *pass = STAPSK;

const char certForum [] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIF3jCCA8agAwIBAgIQAf1tMPyjylGoG7xkDjUDLTANBgkqhkiG9w0BAQwFADCB
iDELMAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0pl
cnNleSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNV
BAMTJVVTRVJUcnVzdCBSU0EgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTAw
MjAxMDAwMDAwWhcNMzgwMTE4MjM1OTU5WjCBiDELMAkGA1UEBhMCVVMxEzARBgNV
BAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0plcnNleSBDaXR5MR4wHAYDVQQKExVU
aGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNVBAMTJVVTRVJUcnVzdCBSU0EgQ2Vy
dGlmaWNhdGlvbiBBdXRob3JpdHkwggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIK
AoICAQCAEmUXNg7D2wiz0KxXDXbtzSfTTK1Qg2HiqiBNCS1kCdzOiZ/MPans9s/B
3PHTsdZ7NygRK0faOca8Ohm0X6a9fZ2jY0K2dvKpOyuR+OJv0OwWIJAJPuLodMkY
tJHUYmTbf6MG8YgYapAiPLz+E/CHFHv25B+O1ORRxhFnRghRy4YUVD+8M/5+bJz/
Fp0YvVGONaanZshyZ9shZrHUm3gDwFA66Mzw3LyeTP6vBZY1H1dat//O+T23LLb2
VN3I5xI6Ta5MirdcmrS3ID3KfyI0rn47aGYBROcBTkZTmzNg95S+UzeQc0PzMsNT
79uq/nROacdrjGCT3sTHDN/hMq7MkztReJVni+49Vv4M0GkPGw/zJSZrM233bkf6
c0Plfg6lZrEpfDKEY1WJxA3Bk1QwGROs0303p+tdOmw1XNtB1xLaqUkL39iAigmT
Yo61Zs8liM2EuLE/pDkP2QKe6xJMlXzzawWpXhaDzLhn4ugTncxbgtNMs+1b/97l
c6wjOy0AvzVVdAlJ2ElYGn+SNuZRkg7zJn0cTRe8yexDJtC/QV9AqURE9JnnV4ee
UB9XVKg+/XRjL7FQZQnmWEIuQxpMtPAlR1n6BB6T1CZGSlCBst6+eLf8ZxXhyVeE
Hg9j1uliutZfVS7qXMYoCAQlObgOK6nyTJccBz8NUvXt7y+CDwIDAQABo0IwQDAd
BgNVHQ4EFgQUU3m/WqorSs9UgOHYm8Cd8rIDZsswDgYDVR0PAQH/BAQDAgEGMA8G
A1UdEwEB/wQFMAMBAf8wDQYJKoZIhvcNAQEMBQADggIBAFzUfA3P9wF9QZllDHPF
Up/L+M+ZBn8b2kMVn54CVVeWFPFSPCeHlCjtHzoBN6J2/FNQwISbxmtOuowhT6KO
VWKR82kV2LyI48SqC/3vqOlLVSoGIG1VeCkZ7l8wXEskEVX/JJpuXior7gtNn3/3
ATiUFJVDBwn7YKnuHKsSjKCaXqeYalltiz8I+8jRRa8YFWSQEg9zKC7F4iRO/Fjs
8PRF/iKz6y+O0tlFYQXBl2+odnKPi4w2r78NBc5xjeambx9spnFixdjQg3IM8WcR
iQycE0xyNN+81XHfqnHd4blsjDwSXWXavVcStkNr/+XeTWYRUc+ZruwXtuhxkYze
Sf7dNXGiFSeUHM9h4ya7b6NnJSFd5t0dCy5oGzuCr+yDZ4XUmFF0sbmZgIn/f3gZ
XHlKYC6SQK5MNyosycdiyA5d9zZbyuAlJQG03RoHnHcAP9Dc1ew91Pq7P8yF1m9/
qS3fuQL39ZeatTXaw2ewh0qpKJ4jjv9cJ2vhsE/zB+4ALtRZh8tSQZXq9EfX7mRB
VXyNWQKV3WKdwrnuWih0hKWbt5DHDAff9Yk2dDLWKMGwsAvgnEzDHNb842m1R0aB
L6KCq9NjRHDEjf8tM7qtj3u1cIiuPhnPQCjY/MiQu12ZIvVS5ljFH4gxQ+6IHdfG
jjxDah2nGN59PRbxYvnKkKj9
-----END CERTIFICATE-----
)EOF";

void setup() {
  char stack; stack_start=&stack; // About to connect: free-heap=41712, stack-used=4294967292/4096
  Serial.begin(115200); delay(3000);
  Serial.printf("\nConnecting to %s\n", ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {delay(500); Serial.print(".");}
  Serial.print("\nConnected, IP Address: ");
  Serial.println(WiFi.localIP());

  // Set up time to allow for certificate validation
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print("Waiting for NTP time sync: ");
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {delay(500); Serial.print("."); now = time(nullptr);}
  struct tm timeinfo;
  gmtime_r(&now, &timeinfo);
  Serial.print("\nCurrent time: ");
  Serial.print(asctime(&timeinfo));
}

void loop() {
  BearSSL::WiFiClientSecure client;
  BearSSL::X509List cert(certForum);
  Serial.printf("About to setTrustAnchors: free-heap=%u, stack-used=%u/4096\n", ESP.getFreeHeap(), stack_size());
  client.setTrustAnchors(&cert);
  Serial.printf("About to connect:         free-heap=%u, stack-used=%u/4096\n", ESP.getFreeHeap(), stack_size());
  int result=client.connect(smtp_address, smtp_port_secure);
  Serial.printf("\nresult=%s\n", (result==1) ? "Success" : "Fail");
  client.stop();
  delay(10000);
}
```

### Debug Messages

```
About to setTrustAnchors: free-heap=41648, stack-used=28/4096
About to connect:         free-heap=41648, stack-used=28/4096
BSSL:_connectSSL: start connection // Debug Level: "SSL"

To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Stack overflow detected

>>>stack>>>

ctx: bearssl
sp: 3fff08e8 end: 3fff14f0 offset: 0000
3fff08e8:  00000001 40100184 46051178 4aef3156...  

------------------------------------------------------------------------------------------------------------------------------

Decoding stack results
0x40100184: ets_post(uint8, ETSSignal, ETSParam) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 244
0x402054a5: __yield() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 194
0x40229ab1: run_code at /home/earle/src/esp-quick-toolchain/arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 107
0x4022a005: point_mul at src/ec/ec_prime_i15.c line 589
0x4022a314: api_mul at src/ec/ec_prime_i15.c line 743
0x4021f6bb: make_pms_ecdh at src/ssl/ssl_hs_client.c line 316
0x4021fff0: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 1295
0x4021e8b4: jump_handshake at src/inner.h line 2211
0x4021ecb6: br_ssl_engine_sendrec_ack at src/ssl/ssl_engine.c line 1168
0x40203d72: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 585
0x40203dc0: BearSSL::WiFiClientSecureCtx::_wait_for_handshake() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 608
0x40203faa: BearSSL::WiFiClientSecureCtx::_connectSSL(char const*) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 1193
0x4020312b: WiFiClient::connect(IPAddress, unsigned short) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266/coredecls.h line 69
0x4020404d: BearSSL::WiFiClientSecureCtx::connect(char const*, unsigned short) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 228
0x4020130c: loop() at C:\Users\r1\Documents\Arduino\ESP8266_BearSSL_Sessions_v3/ESP8266_BearSSL_Sessions_v3.ino line 125
0x40205648: loop_wrapper() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 264
```

brenonevs, 2025-01-13:
a

Rob58329, 2025-01-21:
PS. Still crashes if you change the "`const char certForum2 [] PROGMEM = R"EOF(`" to just "`const char certForum2 [] = R"EOF(`"                 but with slightly different output:

```
Current time: Tue Jan 21 19:50:03 2025
About to setTrustAnchors: free-heap=39552, stack-used=28/4096
About to connect:         free-heap=39552, stack-used=28/4096

To make this dump useful, DECODE IT - https://tinyurl.com/8266dcdr
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Stack overflow detected

>>>stack>>>

ctx: bearssl
sp: 3fff1158 end: 3fff1d60 offset: 0000
3fff1158:  00000001 40100184 46051178 4aef3156  
3fff1168:  3ffe8648 40205491 3fff1224 fffffffc...  
```

------------------------------------------------------------------------------------------------------------------------------

```
Decoding stack results
0x40100184: ets_post(uint8, ETSSignal, ETSParam) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 244
0x40205491: __yield() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 194
0x40229a9d: run_code at /home/earle/src/esp-quick-toolchain/arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 107
0x40229ff1: point_mul at src/ec/ec_prime_i15.c line 589
0x4022a300: api_mul at src/ec/ec_prime_i15.c line 743
0x4021f6a7: make_pms_ecdh at src/ssl/ssl_hs_client.c line 316
0x4021ffdc: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 1295
0x4021e8a0: jump_handshake at src/inner.h line 2211
0x4021eca2: br_ssl_engine_sendrec_ack at src/ssl/ssl_engine.c line 1168
0x40203d86: BearSSL::WiFiClientSecureCtx::_run_until(unsigned int, bool) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 585
0x40203dd4: BearSSL::WiFiClientSecureCtx::_wait_for_handshake() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 608
0x40203fbe: BearSSL::WiFiClientSecureCtx::_connectSSL(char const*) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 1193
0x4020313f: WiFiClient::connect(IPAddress, unsigned short) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266/coredecls.h line 69
0x40204061: BearSSL::WiFiClientSecureCtx::connect(char const*, unsigned short) at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 228
0x40201318: loop() at C:\Users\r1\Documents\Arduino\ESP8266_BearSSL_Sessions_v3/ESP8266_BearSSL_Sessions_v3.ino line 128
0x40205634: loop_wrapper() at C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266\cores\esp8266\core_esp8266_main.cpp line 264

```

mcspr, 2025-01-21:
@Rob58329 pretty sure you don't have https://github.com/esp8266/Arduino/compare/master...mcspr:esp8266-Arduino:bearssl/build-2024_08_09T13_23_00_00 checked out at `C:\ArduinoIDE1819_v9\arduino-1.8.19\hardware\esp8266\esp8266`?
I have c/p the code above and `git-switch`'ed to my local clone of the builder branch. Uploaded to the device, it connects and the final line is `result=Success`

PR #9224 (2025-01-24): Stack thunk yield() fixes
Resolve #9170
Resolve #9218 

See https://github.com/earlephilhower/bearssl-esp8266/pull/19

**upd** [from the original Sep 2024 version](https://github.com/mcspr/esp8266-Arduino/tree/bearssl/build-2024_08_09T13_23_00_00)
can_yield() + yield() -> optimistic_yield(10000) (aka 10ms) for a shorter func body
closed 2025-01-22 by Rob58329