esp8266/Arduino #9233: Compilation error: variable or field 'xxx' declared void

warez4me, 2025-02-27:
When trying to use the type-alias for the structure, I encountered a compilation error.
After an hour of headache, a very strange solution to the problem was found.
I can't believe this is a compiler bug, because this technique is one of the most basic.
I must have made a mistake somewhere, but I can't figure out exactly where.
The "solution" found is to place an empty comment almost anywhere in the parameter list in the function header.

Here error:

[image]

Here Ok ?! 

[image]

And here...

[image]

And here...

[image]

warez4me, 2025-02-27:
I accidentally(?) found the answer to my problem.  Not the same, but similar situation:
[https://arduino.ru/forum/programmirovanie/pochemu-ne-rabotaet-typedef-enum?ysclid=m7n3wda9rj979144486](url)
In short, the arduino preprocessor collects the full source code of the program for the compiler and puts the function definition headers before the definitions of user types. It leads to this error.
The correct way out of this situation is to declare the types in a separate header file.

mcspr, 2025-02-27:
You'd have to ask here instead -> https://github.com/arduino/arduino-cli/issues, we don't control this part of the compilation process

---

Looking at the log, you might have noticed 'ctags' executing under the hood. What it does is parse the .ino, so it can generate an appropriate .cpp file which GCC could actually build. As opposed to the .ino, .cpp requires a certain order of things and format and is pretty strict in regards to breaking the language rules. arduino-cli uses a fairly old way of parsing C++ sources and could make some mistakes in the process.

Check out the temporary directory from the log for more (`%LOCALAPPDATA%\arduino\sketches\$HASHSTRING$`)
e.g. this is the file from the first screenshot, which would have the .cpp extension
```cpp
#include <Arduino.h>
#line 1 "/home/dev/arduino9233/arduino9233.ino"
#line 1 "/home/dev/arduino9233/arduino9233.ino"
void setup();
#line 7 "/home/dev/arduino9233/arduino9233.ino"
void get_bug(my_t* prm); // <--- GCC has no idea what my_t is
#line 12 "/home/dev/arduino9233/arduino9233.ino"
void loop();
#line 1 "/home/dev/arduino9233/arduino9233.ino"
void setup() {
  }

typedef struct { int x, y;} my_t; // <--- first time it is mentioned
my_t bug[2] = {{0,0},{1,1}};

void get_bug(my_t* prm) {
  my_t* ptr;
  ptr = prm;
  }

void loop() {
  get_bug(bug);
  }

```
You can circumvent the issue by using a .h which declares the struct, `#include` usually stays at the top and is not moved when .ino -> .cpp conversion happens
Another way is to use .cpp in the sketch, avoiding .ino -> .cpp completely

(btw, this is not C as well. `struct my_t { int x,y; };` is the equivalent syntax for C++, so you are declaring an actual struct and not an alias for anonymous struct)

closed 2025-02-27 by mcspr