esp8266/Arduino #9234: esp8266, openwrt can't connect

johndoe71rus, 2025-02-27:
trying to connect esp8266 to access point. simple sketch.
```
#include <ESP8266WiFi.h>
#define WIFI_SSID "My_SSID"
#define WIFI_PASS "My_pass"
void setup() {
  Serial.begin(115200);
  WiFi.setPhyMode(WIFI_PHY_MODE_11G);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("Connected");
  Serial.println(WiFi.localIP());
}
void loop() {
}
```
router log only one entry
...
daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: authenticated
daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: authenticated
daemon.info hostapd: wlan1: STA yy:yy:yy:yy:yy:yy IEEE 802.11: authenticated
...
esp does not receive ip address.
```
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
scandone
.wifi evt: 2
....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
......scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
...wifi evt: 7
wifi evt: 7
...scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect        
```
i don't know if this is an esp or AP problem

mcspr, 2025-02-27:
Duplicate of https://github.com/esp8266/Arduino/issues/7965 (and some other ones)

You got the main gist of forcing G mode, iirc it is the main reason
Another thing is forced SDK:3.0.0-dev, which we don't recommend using

> SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687

Please update to Core 3.1.2
When trying out different SDK versions, make sure to erase SDK before flashing the .bin

closed 2025-02-27 by mcspr

johndoe71rus, 2025-03-02:
initially I tried the latest version of SDK. then I found somewhere that after 2.5 many people stopped connecting, so I switched to it.
now I have the correct version of Core: 3.1.2?
```
SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
fpm close 1 
mode : sta(a4:cf:12:f5:22:9d)
add if0
wifi evt: 8
.wifi evt: 2
....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
.....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
......scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
......scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (1f)
wifi evt: 1
STA disconnect: 203
..reconnect
```
wifi AP log 
```
Sun Mar  2 14:12:03 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
Sun Mar  2 14:12:07 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
Sun Mar  2 14:12:11 2025 daemon.notice hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: did not acknowledge authentication response
Sun Mar  2 14:12:15 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
Sun Mar  2 14:12:19 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
Sun Mar  2 14:12:22 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
Sun Mar  2 14:12:26 2025 daemon.info hostapd: wlan1: STA a4:cf:12:f5:22:9d IEEE 802.11: authenticated
```

mcspr, 2025-03-02:
Not much to go on, really.
What is the AP hardware?
What about /etc/config/wireless?

You can't use WPA3, in case you are forcing it. Only WPA2 2.4GHZ works.
`WiFi.setPhyMode(WIFI_PHY_MODE_11G);` is sometimes there to force the AP, but it may work as-is too
(as it does with e.g. various mediatek routers)

---

Just to clarify some terminology
- Core aka esp8266/Arduino is where we are rn, https://github.com/esp8266/Arduino
- SDK aka Espressif NONOS is a separate thing, https://github.com/espressif/ESP8266_NONOS_SDK

NONOS SDK is a closed-source blob from Espressif, WiFi and low level networking is handled there.
We do have a build option for both Arduino IDE and PlatformIO to switch to a different SDK version, in case it broke something. See menu options in Arduino IDE and these build flags for PIO - https://docs.platformio.org/en/latest/platforms/espressif8266.html#sdk-version

Why I mentioned this - 2.4.2 -> 2.5.0 switch switched from SDK:2.2.1 to SDK:3.0.0-dev *by default*. 2.6.0+ switched back, 2.7.0+ ship with SDK:2.2.2(from various git dates).

If you had it working before 2.5.0, swap to SDK:2.2.1 and try again (or any other available version)

johndoe71rus, 2025-03-02:
I used nodemcu and nexx wt1520 openwrt 15 before... .

Now I changed the hardware to x86 openwrt OpenWrt 23.05. 5
I couldn't connect to nodemcu. Now I'm trying a simple sketch in Arduino IDE.

hardware AP
USB device found, idVendor=0b05, idProduct=1784, bcdDevice= 1.01
```
ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3071, rev 021c detected
ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0008 detected
ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
usbcore: registered new interface driver rt2800usb
....
ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
```
cat /etc/config/wireless
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:1d.7/usb2/2-1/2-1:1.0'
	option cell_density '1'
	option band '2g'
	option channel '6'
	option legacy_rates '1'
	option htmode 'HT20'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'My_SSID'
	option encryption 'psk2+ccmp'
	option key 'My_password'
	option ifname 'wlan1'
	option ieee80211w '2'
	option wpa_disable_eapol_key_retries '1'
	option wmm '0'
```

mcspr, 2025-03-02:
> option encryption 'psk2+ccmp'
> option ieee80211w '2'

No mfp support for NONOS, ieee80211w 'required' likely the cause
Don't really remember much about ccmp forcing.

johndoe71rus, 2025-03-02:
i try WPA-PSK/WPA2-PSK Mixed and TKIP CCMP without success.

johndoe71rus, 2025-03-09:
fixed

```
# ieee80211w: whether management frame protection is enabled
# 0 = disabled (default)
# 1 = optional
# 2 = required
```
set AP option ieee80211w '1' and all work