esp8266/Arduino #9243: Blink example doesn't boot on ESP8266mod

lcarlier, 2025-05-20:
Hello,

I have an ESP8266MOD 

[image]

and I an have issue with using the Arduino IDE and the ESP8266 plugin.
I'm now trying to load the LED blink example but it doesn't go further than this on the serial
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00040d90
~ld
```

I'm using the "Generic ESP8266 Module" board in the Arduino IDE and flashing seems to be ok
```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:d1:87:51
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0340
Compressed 265616 bytes to 195728...
Writing at 0x00000000... (8 %)
Writing at 0x00004000... (16 %)
Writing at 0x00008000... (25 %)
Writing at 0x0000c000... (33 %)
Writing at 0x00010000... (41 %)
Writing at 0x00014000... (50 %)
Writing at 0x00018000... (58 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (75 %)
Writing at 0x00024000... (83 %)
Writing at 0x00028000... (91 %)
Writing at 0x0002c000... (100 %)
Wrote 265616 bytes (195728 compressed) at 0x00000000 in 17.4 seconds (effective 122.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

But when I set my GPIO0 to high and start my ESP8266 the output doesn't go further as I'm showing above.

As an experiment, I tried to load the ESP8266 RTOS SDK hello world example from espressif (https://github.com/espressif/ESP8266_RTOS_SDK) and it works fine. I can boot my board without problem.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 7544, room 16 
tail 8
chksum 0x18
load 0x3ffe8408, len 24, room 0 
tail 8
chksum 0x91
load 0x3ffe8420, len 3476, room 0 
tail 4
chksum 0xc1
csum 0xc1
I (43) boot: ESP-IDF v3.4-110-gd412ac60-dirty 2nd stage bootloader
I (43) boot: compile time 18:15:17
I (44) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : QIO
I (65) boot: SPI Flash Size : 2MB
I (71) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (99) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (111) boot:  2 factory          factory app      00 00 00010000 000f0000
I (122) boot: End of partition table
I (129) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x1cc5c (117852) map
0x40210010: _stext at ??:?

I (182) esp_image: segment 1: paddr=0x0002cc74 vaddr=0x4022cc6c size=0x06e88 ( 28296) map
I (192) esp_image: segment 2: paddr=0x00033b04 vaddr=0x3ffe8000 size=0x00544 (  1348) load
I (193) esp_image: segment 3: paddr=0x00034050 vaddr=0x40100000 size=0x00080 (   128) load
I (207) esp_image: segment 4: paddr=0x000340d8 vaddr=0x40100080 size=0x0512c ( 20780) load
I (227) boot: Loaded app from partition at offset 0x10000
Hello world!
This is ESP8266 chip with 1 CPU cores, WiFi, silicon revision 1, 2MB external flash
Restarting in 10 seconds...
```

Since the SDK from espressif is working fine I assume that my hardware setup is ok.

Can anyone tell me what can go wrong?
What can I try to make my ESP8266 work also with arduino IDE?

If there are any information needed please tell me and I will add.

mcspr, 2025-05-20:
Arduino
>  ets Jan  8 2013,rst cause:2, boot mode:(3,0)

RTOS
>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)

Boot mode is different
https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html#id9

mcspr, 2025-05-20:
> But when I set my GPIO0 to high and start my ESP8266 the output doesn't go further as I'm showing above.

...and since there is `~ld`, I would assume that the app successfully loaded. there is no fancy debugging banner like RTOS, just the messages after `ets`
https://github.com/esp8266/Arduino/blob/2c72e6f015f799980593f888c945dc07582da618/bootloaders/eboot/eboot.c#L267
https://github.com/esp8266/Arduino/blob/2c72e6f015f799980593f888c945dc07582da618/bootloaders/eboot/eboot.c#L302-L307

lcarlier, 2025-05-21:
> Arduino
> 
> > ets Jan  8 2013,rst cause:2, boot mode:(3,0)
> 
> RTOS
> 
> > ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> Boot mode is different https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html#id9 https://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-messages-and-modes (_edit_)

Can you explain what 0 and 7 mean? In the boot mode? When I change the level of GPIO0, 2 and 15, only the digit before the coma I.E 3 in my output, seems to change.

Also I noticed in one of the links you shared that they are using a 470uf capacitor and now my system is very stable (at least for the quick tests I did)

So my issue is solved but I still would like to understand that last digit in boot mode(3, y). Do you know

mcspr, 2025-05-22:
> Can you explain what 0 and 7 mean? In the boot mode? When I change the level of GPIO0, 2 and 15, only the digit before the coma I.E 3 in my output, seems to change.
>
> Also I noticed in one of the [...]

Boot process before the 'ets' is part of the ROM and undocumented, so only I could offer only a speculation as to why it does what it does.

1st and 2nd numbers are results of probing the same register
https://github.com/esp8266/Arduino/blob/92002ece2e813047f60538c9e5c71eff0b116810/cores/esp8266/esp8266_peri.h#L54
1st number is focused on one part and bootloader mode selection, 2nd is on the other 'something' and seemingly unused. But in general, if power is unstable or some pins are connected incorrectly, there are discrepancies where there should not be any.

closed 2025-05-22 by mcspr