esp8266/Arduino #9286: WiFiClientSecure fails with -1 but WiFiClient works

MarcelRobitaille, 2025-09-20:
I'm experiencing a really weird problem where my project that used to work now gives error -1 for every network request. I tried switching to unsecure HTTP, and it behaves the same way.

I simplified the example to only connect to my gateway on port 80 and print the status code. With `WiFiClient` this works, but with `WiFiClientSecure` it fails with -1.

This is reproducible for my gateway IP and the 2 APIs the project normally uses. Fetching these with `curl` on my computer on the same network works fine.

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>
#include "secrets.h"

HTTPClient http;

// WiFiClient client;             // <- Works, but can't make HTTPS requests obviously
BearSSL::WiFiClientSecure client; // <- Doesn't work

void setup() {
	Serial.begin(115200);

	WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

	while (WiFi.status() != WL_CONNECTED)
	{
		delay(500);
		Serial.print(".");
	}

	delay(5000);

	client.setInsecure();
	http.begin(client, "192.168.3.1", 80, "");
	int httpCode = http.GET();
	Serial.printf("Status code %u\n", httpCode);
	http.end();
}

void loop() {
}

```

mcspr, 2025-09-21:
> `http.begin(client, "192.168.3.1", 80, "");`

try using a different port? if curl https://... works, use 443 instead of 80

closed 2025-09-21 by mcspr

MarcelRobitaille, 2025-09-21:
I tried that. Like I said, this broke in an existing project using 2 different APIs over HTTPS using fingerprints. I tried to simplify the example as much as possible for the bug report, therefore I used something simpler with port 80.

I also wanted to show that WiFiClient works, so the problem is definitely in BearSSL::WiFiClientSecure, and that the problem is reproducible even for HTTP. Can't BearSSL::WiFiClientSecure make HTTP requests to port 80?

mcspr, 2025-09-21:
> I also wanted to show that WiFiClient works, so the problem is definitely in BearSSL::WiFiClientSecure, and that the problem is reproducible even for HTTP. Can't BearSSL::WiFiClientSecure make HTTP requests to port 80?

nope. WiFiClient -> http, WiFiClientSecure -> https. usually this is port-dependent, why I mentioned the change above

see our debugging options and monitor activity using serial connection. I hope it is understandable, I can't guess my way through the issue here. more and better info about the specifics, better the response is :) http client is limited to the protocol error output, the problem is on the lower level as it seems

MarcelRobitaille, 2025-09-24:
It's somehow working again. Sorry for wasting your time.

Also sorry that I made a bad bug report. I thought I was making a good minimal reproducible example, but of course using the secure client on port 80 doesn't work